WPI-AIM/ambf #154: Build error: /usr/bin/ld: cannot find -lusb-1.0

mr3albert, 2022-01-05:
Cmakes works fine without error.
make seems not able to find -lusb-1.0
```
[ 23%] Built target bullet
[ 23%] Built target adf_loader
[ 71%] Built target chai3d
[ 71%] Built target sensor_msgs_generate_messages_cpp
[ 71%] Built target rosgraph_msgs_generate_messages_eus
[ 71%] Built target roscpp_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_eus
[ 71%] Built target roscpp_generate_messages_cpp
[ 71%] Built target roscpp_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_lisp
[ 71%] Built target std_msgs_generate_messages_lisp
[ 71%] Built target roscpp_generate_messages_lisp
[ 71%] Built target rosgraph_msgs_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_cpp
[ 71%] Built target std_msgs_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_cpp
[ 71%] Built target roscpp_generate_messages_eus
[ 71%] Built target geometry_msgs_generate_messages_nodejs
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_WorldCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_SensorCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ActuatorCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_SensorState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_LightCmd
[ 71%] Built target geometry_msgs_generate_messages_eus
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_VehicleState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_CameraState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_RigidBodyCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ActuatorState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_CameraCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_LightState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_RigidBodyState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ObjectCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_VehicleCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ObjectState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_WorldState
[ 74%] Built target ambf_msgs_generate_messages_eus
[ 74%] Built target geometry_msgs_generate_messages_cpp
[ 74%] Built target geometry_msgs_generate_messages_lisp
[ 76%] Built target ambf_msgs_generate_messages_lisp
[ 76%] Built target geometry_msgs_generate_messages_py
[ 78%] Built target ambf_msgs_generate_messages_py
[ 81%] Built target ambf_msgs_generate_messages_nodejs
[ 83%] Built target ambf_msgs_generate_messages_cpp
[ 83%] Built target sensor_msgs_generate_messages_eus
[ 83%] Built target sensor_msgs_generate_messages_py
[ 83%] Built target sensor_msgs_generate_messages_nodejs
[ 83%] Built target sensor_msgs_generate_messages_lisp
[ 84%] Built target ambf_framework
[ 84%] Built target ambf_msgs_generate_messages
[ 84%] Built target tf_generate_messages_lisp
[ 84%] Built target tf2_msgs_generate_messages_eus
[ 84%] Built target actionlib_generate_messages_lisp
[ 84%] Built target actionlib_msgs_generate_messages_eus
[ 84%] Built target tf2_msgs_generate_messages_lisp
[ 84%] Built target tf_generate_messages_py
[ 84%] Built target tf_generate_messages_nodejs
[ 84%] Built target tf_generate_messages_cpp
[ 84%] Built target actionlib_generate_messages_py
[ 84%] Built target actionlib_msgs_generate_messages_nodejs
[ 84%] Built target actionlib_generate_messages_nodejs
[ 84%] Built target actionlib_generate_messages_eus
[ 84%] Built target actionlib_msgs_generate_messages_cpp
[ 84%] Built target actionlib_msgs_generate_messages_lisp
[ 84%] Built target actionlib_generate_messages_cpp
[ 84%] Built target actionlib_msgs_generate_messages_py
[ 84%] Built target tf_generate_messages_eus
[ 84%] Built target tf2_msgs_generate_messages_cpp
[ 84%] Built target tf2_msgs_generate_messages_nodejs
[ 84%] Built target tf2_msgs_generate_messages_py
[ 87%] Built target ambf_obj
[ 88%] Built target ambf_world
[ 91%] Built target ambf_client_cpp
[ 92%] Built target interpolate_test
[ 93%] Built target dvrk_arm
[ 93%] Built target dvrk_arm_test
[ 95%] Built target glfw
[ 95%] Built target loader_test
[ 95%] Built target math_test
[ 95%] Linking CXX executable ../../bin/lin-x86_64/ambf_simulator
/usr/bin/ld: cannot find -lusb-1.0
/usr/bin/ld: cannot find -lusb-1.0
collect2: error: ld returned 1 exit status
make[2]: *** [ambf_simulator/CMakeFiles/ambf_simulator.dir/build.make:304: ../bin/lin-x86_64/ambf_simulator] Error 1
make[1]: *** [CMakeFiles/Makefile2:3394: ambf_simulator/CMakeFiles/ambf_simulator.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

```

adnanmunawar, 2022-01-05:
What’s the OS and system architecture?

mr3albert, 2022-01-05:
I am using Ubuntu 20.04, ROS noetic. I'm not sure about what you mean by the system architecture. I'm using a Thinkpad X1 carbon that is running a intel core i7 8th gen.

adnanmunawar, 2022-01-05:
Thank you for that information. Since your processor is an Intel i7, I would guess that it's based on the x86_64 architecture. Can you run this command in your terminal and report the output.
```
apt list --installed | grep libusb
```

mr3albert, 2022-01-05:
```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libusb-0.1-4/focal,now 2:0.1.12-32 amd64 [installed]
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64 [installed]
libusbmuxd4/now 1.1.0~git20171206.c724e70f-0.1 amd64 [installed,local]
libusbmuxd6/focal,now 2.0.1-2 amd64 [installed,automatic]
libusbredirparser1/focal,now 0.8.0-1 amd64 [installed,automatic]

```

I looked up a command to install libusb on ubuntu. I used the following command and was able to build successfully. 

`sudo apt-get install libusb-1.0-0-dev`

After installing libusb-1.0-0-dev, I ran 

`apt list --installed | grep libusb`
again 
```
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libusb-0.1-4/focal,now 2:0.1.12-32 amd64 [installed]
libusb-1.0-0-dev/focal,now 2:1.0.23-2build1 amd64 [installed]
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64 [installed]
libusb-1.0-doc/focal,focal,now 2:1.0.23-2build1 all [installed,automatic]
libusbmuxd4/now 1.1.0~git20171206.c724e70f-0.1 amd64 [installed,local]
libusbmuxd6/focal,now 2.0.1-2 amd64 [installed,automatic]
libusbredirparser1/focal,now 0.8.0-1 amd64 [installed,automatic]
```

I'm curious because it seems like I already had libusb installed before I reinstalled it. Is there a specific version of libusb AMBF was looking for?:

adnanmunawar, 2022-01-05:
While using the *-dev version of a pkg is useful when compiling custom applications, such as AMBF, the non "dev" version of libusb-1.0 links fine in my experience. I am curious as to why the linker wasn't able to find the libusb-1.0 before you installed libusb-1.0-0-dev. Can you report the output of this command:

```
dpkg -L libusb-1.0-0
```
And this

```
dpkg -L libusb-1.0-0-dev
```
and finally

```
cd ambf/bin/lin-x86_64
ldd ambf_simulator | grep libusb
```

mr3albert, 2022-01-05:
`dpkg -L libusb-1.0-0
`
output:
```
/.
/lib
/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu/libusb-1.0.so.0.2.0
/usr
/usr/share
/usr/share/doc
/usr/share/doc/libusb-1.0-0
/usr/share/doc/libusb-1.0-0/README
/usr/share/doc/libusb-1.0-0/changelog.Debian.gz
/usr/share/doc/libusb-1.0-0/copyright
/lib/x86_64-linux-gnu/libusb-1.0.so.0

```

`dpkg -L libusb-1.0-0-dev`

output:
```
/.
/usr
/usr/include
/usr/include/libusb-1.0
/usr/include/libusb-1.0/libusb.h
/usr/lib
/usr/lib/x86_64-linux-gnu
/usr/lib/x86_64-linux-gnu/libusb-1.0.a
/usr/lib/x86_64-linux-gnu/pkgconfig
/usr/lib/x86_64-linux-gnu/pkgconfig/libusb-1.0.pc
/usr/share
/usr/share/doc
/usr/share/doc/libusb-1.0-0-dev
/usr/share/doc/libusb-1.0-0-dev/copyright
/usr/lib/x86_64-linux-gnu/libusb-1.0.so
/usr/share/doc/libusb-1.0-0-dev/README
/usr/share/doc/libusb-1.0-0-dev/changelog.Debian.gz
```

```
cd ambf/bin/lin-x86_64
ldd ambf_simulator | grep libusb
```

output:
`	libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007f25ad963000)
`

adnanmunawar, 2022-01-05:
Hmm, this is odd, if you look at the output from `ldd ambf_simulator | grep libusb`, `ambf_simulator` is linking to the library from `libusb-1.0-0` and not the `libusb-1.0-0-dev`. I know that you already built AMBF and can hopefully run it, I am curious as to what will happen you remove `libusb-1.0-0-dev` using `apt remove libusb-1.0-0-dev` and then rebuild AMBF.

mr3albert, 2022-01-05:
rebuilt AMBF after removing `libusb-1.0-0-dev`.

```
[ 23%] Built target bullet
[ 23%] Built target adf_loader
[ 71%] Built target chai3d
[ 71%] Built target sensor_msgs_generate_messages_cpp
[ 71%] Built target rosgraph_msgs_generate_messages_eus
[ 71%] Built target roscpp_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_eus
[ 71%] Built target roscpp_generate_messages_cpp
[ 71%] Built target roscpp_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_lisp
[ 71%] Built target std_msgs_generate_messages_lisp
[ 71%] Built target roscpp_generate_messages_lisp
[ 71%] Built target rosgraph_msgs_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_cpp
[ 71%] Built target std_msgs_generate_messages_nodejs
[ 71%] Built target std_msgs_generate_messages_py
[ 71%] Built target rosgraph_msgs_generate_messages_cpp
[ 71%] Built target roscpp_generate_messages_eus
[ 71%] Built target geometry_msgs_generate_messages_nodejs
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_WorldCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_SensorCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ActuatorCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_SensorState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_LightCmd
[ 71%] Built target geometry_msgs_generate_messages_eus
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_VehicleState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_CameraState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_RigidBodyCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ActuatorState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_CameraCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_LightState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_RigidBodyState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ObjectCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_VehicleCmd
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_ObjectState
[ 71%] Built target _ambf_msgs_generate_messages_check_deps_WorldState
[ 74%] Built target ambf_msgs_generate_messages_eus
[ 74%] Built target geometry_msgs_generate_messages_cpp
[ 74%] Built target geometry_msgs_generate_messages_lisp
[ 76%] Built target ambf_msgs_generate_messages_lisp
[ 76%] Built target geometry_msgs_generate_messages_py
[ 78%] Built target ambf_msgs_generate_messages_py
[ 81%] Built target ambf_msgs_generate_messages_nodejs
[ 83%] Built target ambf_msgs_generate_messages_cpp
[ 83%] Built target sensor_msgs_generate_messages_eus
[ 83%] Built target sensor_msgs_generate_messages_py
[ 83%] Built target sensor_msgs_generate_messages_nodejs
[ 83%] Built target sensor_msgs_generate_messages_lisp
[ 84%] Built target ambf_framework
[ 84%] Built target ambf_msgs_generate_messages
[ 84%] Built target tf_generate_messages_lisp
[ 84%] Built target tf2_msgs_generate_messages_eus
[ 84%] Built target actionlib_generate_messages_lisp
[ 84%] Built target actionlib_msgs_generate_messages_eus
[ 84%] Built target tf2_msgs_generate_messages_lisp
[ 84%] Built target tf_generate_messages_py
[ 84%] Built target tf_generate_messages_nodejs
[ 84%] Built target tf_generate_messages_cpp
[ 84%] Built target actionlib_generate_messages_py
[ 84%] Built target actionlib_msgs_generate_messages_nodejs
[ 84%] Built target actionlib_generate_messages_nodejs
[ 84%] Built target actionlib_generate_messages_eus
[ 84%] Built target actionlib_msgs_generate_messages_cpp
[ 84%] Built target actionlib_msgs_generate_messages_lisp
[ 84%] Built target actionlib_generate_messages_cpp
[ 84%] Built target actionlib_msgs_generate_messages_py
[ 84%] Built target tf_generate_messages_eus
[ 84%] Built target tf2_msgs_generate_messages_cpp
[ 84%] Built target tf2_msgs_generate_messages_nodejs
[ 84%] Built target tf2_msgs_generate_messages_py
[ 87%] Built target ambf_obj
[ 88%] Built target ambf_world
[ 91%] Built target ambf_client_cpp
[ 92%] Built target interpolate_test
[ 93%] Built target dvrk_arm
[ 93%] Built target dvrk_arm_test
[ 95%] Built target glfw
[ 95%] Built target loader_test
[ 95%] Built target math_test
[ 95%] Built target ambf_simulator
[ 96%] Built target psm_ik_test
[ 97%] Built target ecm_ik_test
[ 98%] Built target raven_controller
[ 98%] Built target simulator_example
[100%] Built target world_example
[100%] Built target volumetric_drilling
```
I don't know why it works LOL

adnanmunawar, 2022-01-06:
There is a possibility that your linker paths weren't updated, and installing a new lib via apt-get internally called `ldconfig`. This may have allowed the package to find libusb.

closed 2022-01-08 by adnanmunawar