WPI-AIM/ambf #155: set camera to publishing image, cannot find corresponding ros topic

mr3albert, 2022-01-18:
As the title states, I have uncommented `publishing image: True` in `world.yaml`, but I do not see the `/cameras/camera1/ImageData/*` in my list of rostopics.

What I see when I run the ambf simulator. 
```
(base)  mr3albert@CarbonPanda  ~/Documents/UCSD/SRIP_UCSD  ~/ambf_simulator --launch_file surgical_robotics_challenge/launch.yaml -l 0,1,3,4,14,15 -p 120 -t 1 --override_max_comm_freq 120
INFO ! OVERRIDING MAX COMMUNICATION FREQUENCY TO: 120

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Override the default (or specified in 
                                       ADF) world namespace
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  --plugins arg                        Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
                                       they are valid files
  -l [ --load_multibodies ] arg        Index of Multi-Body(ies) to Launch, 
                                       .e.g. -l 1,2,3 will load multibodies at 
                                       indexes 1,2,3. See launch.yaml file

------------------------------------------------------------

INFO! USING SIMULATION SPEED FACTOR OF: 1
WARNING! For File "surgical_robotics_challenge/launch.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "surgical_robotics_challenge/ADF/world/world_stereo.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "surgical_robotics_challenge/ADF/input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! INITIALIZING ROS NODE HANDLE
INFO! Thread Joined: Plane
INFO! Thread Joined: light2
INFO! CAMERA "cameraL" MONITOR NUMBER "1" IS NOT IN RANGE OF AVAILABLE MONITORS "1", USING DEFAULT
INFO! Thread Joined: cameraL
[... 120 lines cut ...]
/ambf/env/cameras/cameraL/Command
/ambf/env/cameras/cameraL/State
/ambf/env/cameras/cameraR/Command
/ambf/env/cameras/cameraR/State
/ambf/env/lights/light2/Command
/ambf/env/lights/light2/State
/ambf/env/psm1/Actuator0/Command
/ambf/env/psm1/Actuator0/State
/ambf/env/psm1/Sensor0/Command
/ambf/env/psm1/Sensor0/State
/ambf/env/psm1/baselink/Command
/ambf/env/psm1/baselink/State
/ambf/env/psm1/toolyawlink/Command
/ambf/env/psm1/toolyawlink/State
/ambf/env/psm2/Actuator0/Command
/ambf/env/psm2/Actuator0/State
/ambf/env/psm2/Sensor0/Command
/ambf/env/psm2/Sensor0/State
/ambf/env/psm2/baselink/Command
/ambf/env/psm2/baselink/State
/ambf/env/psm2/toolyawlink/Command
/ambf/env/psm2/toolyawlink/State
/rosout
/rosout_agg
```

adnanmunawar, 2022-01-18:
Prior to building AMBF, did you have the `cv-bridge` and `image-transport` packages installed as documented here?
https://github.com/WPI-AIM/ambf/wiki/Camera-feed-and-depth-camera

mr3albert, 2022-01-18:
Yes, I have installed `cv-bridge` and `image-transport`. 

```
(base)  ✘ mr3albert@CarbonPanda  ~/ambf/ambf_models/descriptions/world   ambf-2.0 ±  sudo apt install ros-noetic-cv-bridge
[sudo] password for mr3albert: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-noetic-cv-bridge is already the newest version (1.16.0-1focal.20211124.025534).
The following packages were automatically installed and are no longer required:
  libassimp-dev libassimp5 libcurl4-openssl-dev libeigen3-dev libfreeimage3 libjxr0 libogre-1.9-dev
  libogre-1.9.0v5 liborocos-kdl-dev libtinyxml-dev libtinyxml2.6.2v5 liburdfdom-dev
  liburdfdom-headers-dev liburdfdom-model liburdfdom-model-state liburdfdom-sensor liburdfdom-world
  libzzip-0-13 ros-noetic-angles ros-noetic-laser-geometry ros-noetic-map-msgs
  ros-noetic-media-export ros-noetic-nav-msgs ros-noetic-resource-retriever
  ros-noetic-rosconsole-bridge ros-noetic-roswtf ros-noetic-tf ros-noetic-turtle-tf ros-noetic-urdf
  ros-noetic-visualization-msgs
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
(base)  mr3albert@CarbonPanda  ~/ambf/ambf_models/descriptions/world   ambf-2.0 ±  sudo apt install ros-noetic-image-transport
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-noetic-image-transport is already the newest version (1.12.0-1focal.20210922.192223).
The following packages were automatically installed and are no longer required:
  libassimp-dev libassimp5 libcurl4-openssl-dev libeigen3-dev libfreeimage3 libjxr0 libogre-1.9-dev
  libogre-1.9.0v5 liborocos-kdl-dev libtinyxml-dev libtinyxml2.6.2v5 liburdfdom-dev
  liburdfdom-headers-dev liburdfdom-model liburdfdom-model-state liburdfdom-sensor liburdfdom-world
  libzzip-0-13 ros-noetic-angles ros-noetic-laser-geometry ros-noetic-map-msgs
  ros-noetic-media-export ros-noetic-nav-msgs ros-noetic-resource-retriever
  ros-noetic-rosconsole-bridge ros-noetic-roswtf ros-noetic-tf ros-noetic-turtle-tf ros-noetic-urdf
  ros-noetic-visualization-msgs
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

adnanmunawar, 2022-01-18:
Ok great. Quick question, which world file are you setting the `publish image: True` flag in? The [`world.yaml`](https://github.com/collaborative-robotics/surgical_robotics_challenge/blob/master/ADF/world/world_stereo.yaml#L42) file from the `surgical robotics challenge` or the [`world.yaml`](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_models/descriptions/world/world.yaml) file from the main AMBF repo?

mr3albert, 2022-01-18:
It's the `world.yaml` in the main AMBF repo. 

'world.yaml' path:
`/home/mr3albert/ambf/ambf_models/descriptions/world`

adnanmunawar, 2022-01-18:
I see. The `publish image` flag it set for each camera, and if that camera is loaded, its video is then streamed out. It appears from your console output that you are trying to launch the surgical robotics challenge, which has its own `launch_file` which defines a different [`world.yaml`](https://github.com/collaborative-robotics/surgical_robotics_challenge/blob/master/ADF/world/world_stereo.yaml). You will need to set the `publish image` flag for the camera(s) in that world file.

mr3albert, 2022-01-19:
I added the `publish image` flag to the `world_stereo.yaml` as defined in the `launch_file` in `surgical_robotic_challenge`. AND IT WORKED! Thank you so much!

closed 2022-01-19 by mr3albert