WPI-AIM/ambf #156: How to get AMBF Python client to work?

astitwlathe, 2022-02-11:
Hi Everyone,
I need your help to get started on getting AMBF working. I installed AMBF. The simulator is working. However, I am unable to find a way to get AMBF python client to work.  What should I do to get it to work?  It is written that I should not call setup.py in this link: https://github.com/WPI-AIM/ambf/blob/c16127e8d44c57c58026b94e19d3db829ff02e2b/ambf_ros_modules/ambf_client/setup.py. Sorry, I am just getting started on ROS. I would appreciate any help you all can give. Thank you.

adnanmunawar, 2022-02-11:
You do not need to install the Python client yourself, just source the ambf install as mentioned at the end of [Installing AMBF Instructions](https://github.com/WPI-AIM/ambf/wiki/Installing-AMBF)

If you have already done so, have you looked at some of the tutorials in the [Wiki](https://github.com/WPI-AIM/ambf/wiki)?
This [tutorial](https://github.com/WPI-AIM/ambf/wiki/Controlling-dVRK-Manipulators) for example demonstrates launching some robots and controlling them via the Python client.

astitwlathe, 2022-02-11:
Thank you for your reply. I have installed AMBF. The simulator works. I tried this example https://github.com/WPI-AIM/ambf/tree/ambf-2.0/ambf_ros_modules/examples/object_control_example. 

When I executed the command: 
"python control_object.py -o baselink".

I got the error: 
"ImportError: No module named ambf_msgs.msg". 
I am trying to learn AMBF for surgical robotics competition. I am beginner in ROS. Sorry for noob questions.

adnanmunawar, 2022-02-11:
No worries. I am assuming that you are using Linux (Ubuntu) and have ROS already installed. If that is the case, the issue may indicate that "AMBF" has not been "sourced" in the terminal window in which you were running the `python control_object.py -o baselink` command. You can source AMBF with the instructions that I added earlier. Also, depending upon which Ubuntu version you are using (between 18.04 and 20.04), you might want to use the default python interpreter in this command `python control_object.py -o baselink`, i.e. `python3` for Ubuntu 20.04 and just `python` for Ubuntu 18.04

astitwlathe, 2022-02-12:
Hi Adnan,
As suggested, I ran the tutorial [Controlling dVRK Manipulators](https://github.com/WPI-AIM/ambf/wiki/Controlling-dVRK-Manipulators). I could not run "Controlling via AMBF Python Client". I am getting an error when I execute python client.py. The file client.py contains the python code snippet listed in the tutorial. The error is: 
MTM Wrist Fixed Rotation:
Traceback (most recent call last):
  File "client.py", line 42, in <module>
    print(mtm_wrist_handle.get_rpy())

adnanmunawar, 2022-02-22:
Did this resolve your issue?

astitwlathe, 2022-02-22:
Yes, it worked. Thanks.

closed 2022-02-22 by adnanmunawar