WPI-AIM/ambf #157: Build Error

parthnatekar, 2022-02-27:
Hi,

I'm trying to build this on Ubuntu 18.04 using ROS installed via conda.

I'm getting the following build error:

```
[ 24%] Built target bullet
[ 24%] Building CXX object CMakeFiles/adf_loader.dir/adf_loader/adf_loader_base.cpp.o
In file included from <<PATH>>/envs/ros/include/boost/config/detail/select_stdlib_config.hpp:24:0,
                 from<<PATH>>/envs/ros/include/boost/config.hpp:44,
                 from <<PATH>>/envs/ros/include/boost/filesystem/path.hpp:18,
                 from <<PATH>>/ambf/ambf_framework/afPath.h:48,
                 from <<PATH>>/ambf/ambf_framework/afAttributes.h:49,
                 from <<PATH>>/ambf/adf_loader/adf_loader_base.h:48,
                 from <<PATH>>/ambf/adf_loader/adf_loader_base.cpp:43:
<<PATH>>/ambf/ambf_framework/version:44:1: error: ‘MAJOR’ does not name a type
 MAJOR=2
 ^~~~~
CMakeFiles/adf_loader.dir/build.make:75: recipe for target 'CMakeFiles/adf_loader.dir/adf_loader/adf_loader_base.cpp.o' failed
make[2]: *** [CMakeFiles/adf_loader.dir/adf_loader/adf_loader_base.cpp.o] Error 1
CMakeFiles/Makefile2:876: recipe for target 'CMakeFiles/adf_loader.dir/all' failed
make[1]: *** [CMakeFiles/adf_loader.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

adnanmunawar, 2022-02-28:
Thanks for reporting the issue. Can you check if renaming the files called `version` located at [`ambf/ambf_framework/version`](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_framework/version) and [`ambf/external/chai3d/src/version`](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/external/chai3d/src/version) to something different, e.g. `version_bak` resolve this build issue?

parthnatekar, 2022-03-03:
Hi,

Thanks for you reply.

Your solution solves the original issue, but I now have a different issue.

```
<<PATH>>/libroscpp.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
<<PATH>>/librosconsole.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
collect2: error: ld returned 1 exit status
ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/build.make:123: recipe for target '../bin/lin-x86_64/dvrk_arm_test' failed
make[2]: *** [../bin/lin-x86_64/dvrk_arm_test] Error 1
CMakeFiles/Makefile2:2808: recipe for target 'ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all' failed
make[1]: *** [ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

adnanmunawar, 2022-03-04:
Hmm. Can you verify what compiler is being used for compiling AMBF? You can go to AMBF's build folder and run 
```bash
cmake -LA -N . | grep -i CMAKE_CXX_COMPILER:FILEPATH
```
to show the result. I am guessing that it would be GCC, in which case, can you print the version by 

``` bash
`cmake -LA -N . | grep -i CMAKE_CXX_COMPILER:FILEPATH | cut -d'=' -f2-` --version
```
 Regardless, the issue seems to be similar to what is explained [here](https://www.codetd.com/en/article/7839605). I am afraid that it may not be limited to just "DVRK_ARM" package but other packages in the "ambf_ros_modules" folder as well. 

To start, can you try adding this line `add_definitions(-D_GLIBCXX_USE_CXX11_ABI=0)` in the [CMakeLists.txt](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_ros_modules/dvrk_arm/CMakeLists.txt#L48) file of the dvrk_arm package. You can add it to after line 48.

parthnatekar, 2022-03-11:
Hi,

```cmake -LA -N . | grep -i CMAKE_CXX_COMPILER:FILEPATH``` gives me 

```CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++``` which is simlinked to 

```CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++``` which is in turn simlinked to 

```/etc/alternatives/c++ -> /usr/bin/g++```

```cmake -LA -N . | grep -i CMAKE_CXX_COMPILER:FILEPATH | cut -d'=' -f2-` --version``` gives me 

```c++ (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0```

I tried the fix you mentioned, but I get the following build error:

``../../devel/lib/libdvrk_arm.so: undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
../../devel/lib/libdvrk_arm.so: undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
<<PATH>>/lib/librosconsole.so: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
<<PATH>>/lib/libroscpp.so: undefined reference to `std::__cxx11::basic_stringstream<char, std::char_traits<char>, std::allocator<char> >::basic_stringstream()@GLIBCXX_3.4.26'
../../devel/lib/libdvrk_arm.so: undefined reference to `ros::init(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::string const&, unsigned int)'
collect2: error: ld returned 1 exit status
ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/build.make:123: recipe for target '../bin/lin-x86_64/dvrk_arm_test' failed
make[2]: *** [../bin/lin-x86_64/dvrk_arm_test] Error 1
CMakeFiles/Makefile2:2808: recipe for target 'ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all' failed
make[1]: *** [ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

adnanmunawar, 2022-03-14:
Thanks for trying out the suggestion. We can try out two different things from here.
1. Change the CMAKE_CXX_COMPILER to Clang++ and try to compile.
   For this option, assumimg that you have clang installed at `/usr/bin/clang++`. Head to the AMBF build directory.
```
cd ambf/build
cmake -DCMAKE_CXX_COMPILER=/usr/bin/clang++ .
make
```
I am curious to see if this resolves the issue.

2. Alternatively, you can try to update GCC using the instructions [here](https://linuxconfig.org/how-to-switch-between-multiple-gcc-and-g-compiler-versions-on-ubuntu-20-04-lts-focal-fossa) and try to rebuild.

If either of these doesn't work, we can try to set up a zoom / remote session.

parthnatekar, 2022-03-15:
Hi,

Both these solutions get rid of the ```__cxx11``` errors but the ```ros``` reference errors remain:

```
../../devel/lib/libdvrk_arm.so: undefined reference to `ros::init(std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&, std::string const&, unsigned int)'
../../devel/lib/libdvrk_arm.so: undefined reference to `ros::console::initializeLogLocation(ros::console::LogLocation*, std::string const&, ros::console::levels::Level)'
../../devel/lib/libdvrk_arm.so: undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/build.make:123: recipe for target '../bin/lin-x86_64/dvrk_arm_test' failed
make[2]: *** [../bin/lin-x86_64/dvrk_arm_test] Error 1
CMakeFiles/Makefile2:2808: recipe for target 'ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all' failed
make[1]: *** [ambf_ros_modules/dvrk_arm/CMakeFiles/dvrk_arm_test.dir/all] Error 2
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
```

A zoom call would be good, how would we go about that?

adnanmunawar, 2022-03-16:
That's great. Do you still have the line that I recommended adding in this comment:

> To start, can you try adding this line `add_definitions(-D_GLIBCXX_USE_CXX11_ABI=0)` in the [CMakeLists.txt](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_ros_modules/dvrk_arm/CMakeLists.txt#L48) file of the dvrk_arm package. You can add it to after line 48.

If so, can you get rid of it and try to build again?

parthnatekar, 2022-03-16:
Hi,

I was able to complete the build this way (albeit with a lot of warnings).

I am closing this for now - thanks for the help!

Just a comment for others who might read through this - I tried on multiple machines and the version file renaming issue came up on all of them. Changing the file name to something else like ```version_bak``` solved the issue.

adnanmunawar, 2022-03-16:
Glad it was resolved and thanks for the useful information. Let's keep this open for now so that it can be closed with a proper patch.

commit 99244d45 (2022-03-21): Fixes #157. Renaming version files to version.txt
closed 2022-03-21 by adnanmunawar