WPI-AIM/ambf #159: ERROR in running 'input_device_teleop_example'

GPROWIRELESS, 2022-03-15:
Hello,
I'm new on AMBF and I try to run the example in the AMBF files

The example is 'https://github.com/WPI-AIM/ambf/tree/ambf-1.0/ambf_ros_modules/examples/input_device_teleop_example'.
At the second step, after I run: sudo ./ambf_simulator -p60 -t1 --launch_file home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml -l10 -i0
The Error is :

bad file
Failed to Load home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml
ERROR! COULDN'T DETERMINE THE CORRECT ADF LOADER FOR THE FILE "home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml"

My environment:
Ubuntu 20.04
ROS: noetic
version:1.15.14

What's the ADF LOADER and how can I fix it?

Thank you for your help,
Best
FrogsHunter

adnanmunawar, 2022-03-16:
Hello,
Thanks for posting the detail of your steps. In your command:

```bash
sudo ./ambf_simulator -p60 -t1 --launch_file home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml -l10 -i0
```
I think there is a missing `/` before the `home/dvrk/...` which is probably why the ADF loader is complaining. 

Another possible issue that you may end up running into is with the `sudo` command. Is there any specific reason you are using `sudo`?

GPROWIRELESS, 2022-03-16:
Thank you for answering my question! I have fixed the problem, it was caused by this root. I copy the path from file properties and missed something. I typed it by myself and it didn't happen again. I used sudo since the USB device can't be detected without sudo.

GPROWIRELESS, 2022-03-16:
Well, I still have issues with this example. Right now I can run the command:
sudo ./ambf_simulator -p60 -t1 --launch_file /home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml -l0 -i8
but the model can't be controlled by the panel created by python.

GPROWIRELESS, 2022-03-16:
The simulator window shows up but I can't control the model.

adnanmunawar, 2022-03-16:
Can you try the new commit above? Please make sure to set the `-l 10` instead of `-l 0 ` as you have above. Also, this example does not rely on any USB devices, so running with `sudo` is not required.

GPROWIRELESS, 2022-03-16:
I have tried to run with -l 10 but no model showed in the window, so I change it to -I 0.
[image]

GPROWIRELESS, 2022-03-16:
dvrk@dvrk-desktop:~/ambf/bin/lin-x86_64$ ./ambf_simulator -p60 -t1 --launch_file /home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml -l10 -i0

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Override the default (or specified in 
                                       ADF) world namespace
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  --plugins arg                        Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
                                       they are valid files
  -l [ --load_multibodies ] arg        Index of Multi-Body(ies) to Launch, 
                                       .e.g. -l 1,2,3 will load multibodies at 
                                       indexes 1,2,3. See launch.yaml file

------------------------------------------------------------

INFO! USING SIMULATION SPEED FACTOR OF: 1
WARNING! For File "/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/./world/world.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/./input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! INITIALIZING ROS NODE HANDLE
INFO! Thread Joined: Plane
INFO! Thread Joined: light_left
INFO! Thread Joined: camera1
INFO! Thread Joined: World
ERROR! -l 10 greater than total number of files 1. Ignoring 
Found Lib: libdvrk_arm.so
No of dVRK Masters detected: 0
WARNING! FAILED TO LOAD DEVICE: "MTML"

adnanmunawar, 2022-03-16:
Did you run the `gripper_control_via_device_ifc.py` script before running the simulator?

GPROWIRELESS, 2022-03-16:
Yes, I run the command 'python3 gripper_control_via_device_ifc.py' first, then the simulator.

adnanmunawar, 2022-03-16:
Can you copy and paste the contents on these two files:
`ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml` and `ambf/ambf_ros_modules/examples/input_device_teleop_example/input_devices/input_devices.yaml`

GPROWIRELESS, 2022-03-17:
What should I do in those two files? Change the path to the absolute path?

adnanmunawar, 2022-03-17:
No, I am asking if you modified any of these two files in your local copy? Based on your comment above, it appears that you did.

> INFO! USING SIMULATION SPEED FACTOR OF: 1 WARNING! For File "/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/launch.yaml", ADF version not defined thus assuming VERSION_1_0 [...]

GPROWIRELESS, 2022-03-17:
input devices: [
  MTML, #0
  MTMR, #1
  Geomagic Touch, #2
  Razer Hydra0, #3
  Razer Hydra1, #4
  Novint Falcon1, #5
  Novint Falcon2, #6
  Sigma7, #7
  Omega7 #8
  ]

MTML:
    hardware name: MTML
    haptic gain: {linear: 0.03, angular: 0.0}
    workspace scaling: 5
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper small red.yaml"
    location: {
      position: {x: -0.5, y: 0.0, z: 0},
      orientation: {r: 0, p: 0.0, y: 0}}
    button mapping: {
      a1: 1,
      a2: 6,
      next mode: 3,
      prev mode: 4}
    pair cameras: [camera2] # The cameras paired with this IID-SDE pair
    # visible: True
    # visible size: 0.005

MTMR:
  hardware name: MTMR
  haptic gain: {linear: 0.03, angular: 1}
  workspace scaling: 5
  simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper small red.yaml"
  # root link:
  location: {
    position: {x: -0.5, y: 0, z: 0},
    orientation: {r: 0, p: 0, y: 0}}
  button mapping: {
    a1: 1,
    a2: 2,
    next mode: 3,
    prev mode: 4}
  pair cameras: [camera1, camera2] # The motion of the IID-SDE is w.r.t the first camera in the list

Geomagic Touch:
    hardware name: PHANTOM Omni
    haptic gain: {linear: 0.000, angular: 0.0}
    deadband: 0.01
    max force: 2
    max jerk: 0.2
    workspace scaling: 10
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/resistive/sensorized_box/gripper.yaml"
    location: {
      position: {x: 0.0, y: 0.5, z: 0.0},
      orientation: {r: 0, p: 0, y: 0}}
    button mapping: {
      a1: 0,
      a2: -1,
      g1: 1,
      next mode: 2,
      prev mode: 3}
    # visible: True
    # visible size: 0.005

Razer Hydra0:
    hardware name: Razer Hydra
    haptic gain: {linear: 10.0, angular: 0.0}
    # controller gain: {
    #   linear: {P: 0.0, D: 0.0},
    #   angular: {P: 0.0, D: 0.0}}
    workspace scaling: 5
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper.yaml"
    # root link: "/ambf/env/BODY l_wrist_roll_link"
    location: {
      position: {x: 0.0, y: -1.0, z: 0},
      orientation: {r: 0.0, p: 0.0, y: 0.0}}
    # orientation offset: {r: 0.0, p: 0.0, y: 0}
    button mapping: {
      a1: 0,
      a2: 1,
      next mode: 2,
      prev mode: 3}
    # visible: True
    # visible size: 0.005

Razer Hydra1:
    hardware name: Razer Hydra
    haptic gain: {linear: 10.0, angular: 0.0}
    workspace scaling: 5
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper.yaml"
    # root link: "/ambf/env/BODY r_wrist_roll_link"
    location: {
      position: {x: 0.0, y: 0.5, z: 0},
      orientation: {r: 0, p: 0, y: 0}}
    # orientation offset: {r: 0.0, p: 0.0, y: 0}
    button mapping: {
      a1: 0,
      a2: 1,
      next mode: 2,
      prev mode: 3}
    # visible: True
    # visible size: 0.005

Novint Falcon1:
    hardware name: Falcon
    haptic gain: {linear: 0.5, angular: 0.0}
    deadband: 0.01
    max force: 10
    workspace scaling: 10
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper.yaml"
    location: {
      position: {x: 0.0, y: 0.5, z: 0.0},
      orientation: {r: 0, p: 0, y: 0}}
    button mapping: {
      a1: 0,
      a2: 2,
      next mode: 3,
      prev mode: 1}
    # visible: True
    # visible size: 0.005

Novint Falcon2:
    hardware name: Falcon
    haptic gain: {linear: 0.5, angular: 0.0}
    deadband: 0.01
    max force: 10
    workspace scaling: 10
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper.yaml"
    location: {
      position: {x: 0.0, y: -0.5, z: 0.0},
      orientation: {r: 0, p: 0, y: 0}}
    button mapping: {
      a1: 0,
      a2: 2,
      next mode: 3,
      prev mode: 1}
    # visible: True
    # visible size: 0.005

Sigma7:
    hardware name: sigma.7
    haptic gain: {linear: 0.01, angular: 0.0}
    deadband: 0.01
    workspace scaling: 5
    enable joint control: true
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/resistive/sensorized_box/gripper.yaml"
    location: {
      position: {x: 0.5, y: 0.5, z: 0.0},
      orientation: {r: 0.0, p: 0.0, y: 0.0}}
    max force: 2
    # visible: True
    # visible size: 0.005

Omega7:
    hardware name: omega.7
    haptic gain: {linear: 0.01, angular: 0.0}
    deadband: 0.01
    workspace scaling: 5
    enable joint control: true
    simulated multibody: "/home/dvrk/ambf/ambf_models/descriptions/multi-bodies/grippers/pr2 gripper.yaml"
    location: {
      position: {x: 0.5, y: 0.5, z: 0.0},
      orientation: {r: 0.0, p: 0.0, y: 0.0}}
    max force: 2
    # visible: True
    # visible size: 0.005

GPROWIRELESS, 2022-03-17:
I change the absolute path in the file'/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/input_devices'. If I don't do that, there will be ERROR under the WARNING
WARNING! For File "/home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/./input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
bad file
Failed to Load /home/dvrk/ambf/ambf_ros_modules/examples/input_device_teleop_example/./input_devices/../multi-bodies/grippers/pr2 gripper small red.yaml

adnanmunawar, 2022-03-17:
I think we are narrowing in on the problem. You replaced the contents of `ambf/ambf_ros_modules/examples/input_device_teleop_example/input_devices/input_devices.yaml` file from somewhere else so you should have mentioned this earlier. Please restore it to the original version COMPLETELY to run the `input_device_teleop` example. Not only is the device order different in this file, the `simulated multibodies` are also different from what is present in the original [`input_devices.yaml`](https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_ros_modules/examples/input_device_teleop_example/input_devices/input_devices.yaml) file.

GPROWIRELESS, 2022-03-17:
Oh, I'm so sorry. I don't realize that. I replace the input_device.yaml from the pull request. Also, I replace the ambf/ambf_framework/afInputDevices.cpp from the pull request. I completely reinstall the AMBF now. The example works well.

GPROWIRELESS, 2022-03-17:
I was wondering how can I use the haptic device as an input device in AMBF? Do I need to create a python client to get the input data and send it to the AMBF?

GPROWIRELESS, 2022-03-17:
I mean for the example 'input_device_teleop_example', I know I need to add my device name in the input_device.yaml. I also read the pull request, it seems I need to replace the afInputDevices.cpp to handle the '.' in the name of the device. But those two steps are not enough, I have no idea for the next step.

adnanmunawar, 2022-03-17:
Glad to hear that the example is working now. 

What haptic device do want to use with AMBF? And what do you want to do with the haptic device, i.e. 
1. Control a robot (with links connected via joints such that you shall  be controlling the joints to make the robot react to your input device), OR
2. Control a free body, such as a tool or a free-floating gripper that does not require you to operate in joint space.

If its the first option, you may use the Python client OR create an AMBF Plugin where you can do everything you want.

GPROWIRELESS, 2022-03-17:
I try to use Omega.7 haptic device to control DVRK arms such as this one:
[image]

GPROWIRELESS, 2022-03-17:
I have installed the driver for my haptic device, but I don't know how to get data from the driver and send those data to the python client. Can you give me any advice?

adnanmunawar, 2022-03-21:
Let's try to set up some time this Thursday for a zoom meeting as I don't have a force dimension device at hand to test this out.