WPI-AIM/ambf #175: Grasp a soft body with actuators

yannnnnnnng, 2022-05-09:
Hey Adnan,
Thanks in advance for your time.

I'm new on AMBF and I am considering using actuators in the [example](https://github.com/WPI-AIM/ambf/tree/ambf 2.0/ambf_ros_modules/examples/sensors_actuators_example) to grasp some soft bodies (like soft body examples you provided [here](https://github.com/WPI-AIM/ambf/issues/39)). But I found the soft body can't be grasped by the actuators even thouth the end-effector is close to the softbody. Do you know how to grasp a soft body with actuators?
Thank you for your help

adnanmunawar, 2022-05-11:
Hi Yang,
Thanks for reporting this. It is possible that soft body grasping using the combination of sensors and actuators may be broken. I shall look into it.

yannnnnnnng, 2022-06-01:
Hey Adnan,
Thank you for your reply, I'm now working on a project which i need to use a PSM to grasp a soft body tissue, since the actuators doesn't work now, is there any other way to do it?

JohannesSchwaiger, 2022-06-02:
Hey Adnan,
I would also be very interested in grabbing softbodies. Is there a way to do it force based? right now all my grippers slip right through the cloth/softbody when I want to grab it or pull more than just a tiny bit. I tried incresing the vertex count at the grasping point, but that does not change anything. Is there another way to improve force based methods?

Sensor and actuator based would also work for me, if it wasn't broken.

Thanks for your time.

adnanmunawar, 2022-06-16:
Hi Yang and Johannes,
I finally had some time to look at the code and I realized that we never implemented the grasping of soft bodies via ROS interfaces of "sensors" and "actuators". We did have grasping of a softbody (a face or a node/vertex) using the "input device interface" which does not rely on the ROS interfaces and instead uses the internal API. 

A quick background of how this works might be helpful for you both. A sensor can detect a RigidBody and record its name, and how far it is from the sensor's origin. This data can be streamed via the sensors' ROS interface and we can trigger an actuator to grasp the desired RigidBody at the desired offset. On the other hand, a soft body is a little tricky as it requires us to include the data of the face (and maybe the enclosing vertices/nodes) in the sensor's output data. Similarly, for the actuator, we also need to specify this information for grasping since, unlike a rigid body, a soft body must be grasped at specific nodes/vertices.  
 
That being said, I plan on extending the ROS interface of sensors and actuators to allow grasping of a soft body but I will need to think of a way to incorporate it cleanly.

closed 2022-07-25 by yannnnnnnng

adnanmunawar, 2022-08-07:
Please check the [comment](https://github.com/WPI-AIM/ambf/issues/179#issuecomment-1207277654) in Issue #179.