WPI-AIM/ambf #178: namespace arg in simulator executable does not appear to be working

htp2, 2022-06-09:
I was hoping to run multiple simulations simultaneously, but cannot start two as it creates two ros nodes of the same name (/ambf_comm_node). I noticed this was addressed in #54 by adding a --ns arg to the ambf_simulator executable.

However, when I try this, I don't see any change.

<Terminal 1>
```
./ambf_simulator --ns /test/
```
<Terminal 2>
```
rosnode list

/ambf_comm_node
/rosout
```

Curious if this feature was removed, or if there is a better way to run multiple simulations at the same time. Thanks!

adnanmunawar, 2022-06-12:
Thanks for pointing this out Henry. This must have broken quite a while back, I shall fix it soon.

adnanmunawar, 2022-06-16:
Hi Henry, can you please try PR #180 to see if it resolves the issue.

htp2, 2022-06-24:
Hi Adnan - got a chance to test this today. Works great, thanks!

adnanmunawar, 2022-08-01:
Fixed in the above PR

closed 2022-08-01 by adnanmunawar