WPI-AIM/ambf #179: Grasping of Softbodies

JohannesSchwaiger, 2022-06-15:
Hey Adnan,
thanks for your time in advance. 

Is there any possibility to grasp a softbody witha rigid body? My scenario looks like this:

[image]

[image]

I want to grab the yellow rubber band fixed between two blue rigid body pole. The gripper (grey) is also rigid body.

Sensors and Actuators on Softbody don't seem to work. Could something like the pic above be done, where the yellow rubber band is fixed to a green rigid body, so I can then grab the green middle part?

Or is there another way for grasping?

Kind regards,
Johannes

adnanmunawar, 2022-06-17:
Dear Johannes,

Thanks for the wonderful drawing :). In short, it is possible with the "input device interface" which does not use the ROS Communication and only internal methods but not with the ROS topics and therefore the Python Client currently. That being said, I am thinking of adding this in the next few days, I just have to figure out a clean way to do so. Linking this to Issue #175

JohannesSchwaiger, 2022-06-27:
Hey Adnan, that is great to hear!
Thank you for your support

JohannesSchwaiger, 2022-08-05:
Hello Adnan,

were you able to implement the grasping yet? Or would there be a commit where grasping softbody was possible via ros/python? Right now I have Issues slipping through a Rubber band that I created using your softbody examples whenever I exert a little bit of force/pull. Is there a work around for that? I tried increasing vertex density, but that did not change things, I also experimented with primitives, which also did not help.

Kind regards,
Johannes

adnanmunawar, 2022-08-06:
Hi Johannes,
Thanks for the patience. I have a beta implementation of grasping soft bodies on this [remote/branch](https://github.com/adnanmunawar/ambf/tree/feat-softbody-grasping). You'll need to rebuild and re-source AMBF after cloning. Afterward, please check out the updated [`sensing_and_grasping`](https://github.com/adnanmunawar/ambf/tree/feat-softbody-grasping/ambf_ros_modules/examples/sensors_actuators_example) example which is also on that branch.

Here is a demo video. 

https://user-images.githubusercontent.com/5005445/183264950-e9531753-9e4e-4032-b682-eca3d2e9b59b.mp4

JohannesSchwaiger, 2022-08-07:
Hello Adnan,
that is great news. Thank you for your help and assistance. In my first test it seems to work as intended. I will incorporate this in the coming week.
Kind regards,
Johannes

adnanmunawar, 2022-10-24:
Reopening as the changes had not been incorporated in the main repo.