WPI-AIM/ambf #185: Build fails in ambf-2.0 branch

melodysu83, 2022-08-16:
Hi Adnan,
We're recently trying to migrate our commits to the AMBF-2.0 branch but are having trouble building the new code.
Here is the **cmake** output: (which went well without error)
```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Using CATKIN_DEVEL_PREFIX: /home/melody/ambf-new/AMBF-WPI/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.12", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/melody/ambf-new/AMBF-WPI/build/test_results
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.12") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
-- *** FOUND ROS ON THIS MACHINE, ENABLING SUPPORT FOR AMBF_ROS MODULES
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- ambf_msgs: 16 messages, 0 services
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Warning (dev) at external/tf_function/CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    PROJECT_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

-- -- *** FOUND AMBF_COMM, AMBF_MSGS, AMBF_CLIENT and DVRK_ARM, ENABLING DVRK DEVICE SUPPORT
-- -- *** FOUND OPENCV
-- -- *** FOUND ROS_CV_BRIDGE and IMAGE_TRANSPORT, ENABLING VIDEO PUBLISHING SUPPORT
-- -- *** RAZER HYDRA'S ROS PACKAGE NOT FOUND
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   program_options
-- Configuring done
-- Generating done
-- Build files have been written to: /home/melody/ambf-new/AMBF-WPI/build
```

But then when I try: **make ..** , a few errors occurred.
```
[ 12%] Building CXX object CMakeFiles/ambf_framework.dir/ambf_framework/afUtils.cpp.o
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:63:107: error: specialization of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ in different namespace [-fpermissive]
 cQuaternion afUtils::getRotBetweenVectors<cQuaternion, cVector3d>(const cVector3d &v1, const cVector3d &v2){
                                                                                                           ^
In file included from /home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:43:0:
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.h:66:15: error:   from definition of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ [-fpermissive]
     static T1 getRotBetweenVectors(const T2 &v1, const T2 &v2);
               ^
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:98:103: error: specialization of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ in different namespace [-fpermissive]
 cMatrix3d afUtils::getRotBetweenVectors<cMatrix3d, cVector3d>(const cVector3d &v1, const cVector3d &v2){
                                                                                                       ^
In file included from /home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:43:0:
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.h:66:15: error:   from definition of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ [-fpermissive]
     static T1 getRotBetweenVectors(const T2 &v1, const T2 &v2);
               ^
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:113:109: error: specialization of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ in different namespace [-fpermissive]
 btQuaternion afUtils::getRotBetweenVectors<btQuaternion, btVector3>(const btVector3 &v1, const btVector3 &v2){
                                                                                                             ^
In file included from /home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:43:0:
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.h:66:15: error:   from definition of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ [-fpermissive]
     static T1 getRotBetweenVectors(const T2 &v1, const T2 &v2);
               ^
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:132:107: error: specialization of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ in different namespace [-fpermissive]
 btMatrix3x3 afUtils::getRotBetweenVectors<btMatrix3x3, btVector3>(const btVector3 &v1, const btVector3 &v2){
                                                                                                           ^
In file included from /home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.cpp:43:0:
/home/melody/ambf-new/AMBF-WPI/ambf_framework/afUtils.h:66:15: error:   from definition of ‘template<class T1, class T2> static T1 ambf::afUtils::getRotBetweenVectors(const T2&, const T2&)’ [-fpermissive]
     static T1 getRotBetweenVectors(const T2 &v1, const T2 &v2);
               ^
CMakeFiles/ambf_framework.dir/build.make:62: recipe for target 'CMakeFiles/ambf_framework.dir/ambf_framework/afUtils.cpp.o' failed
make[2]: *** [CMakeFiles/ambf_framework.dir/ambf_framework/afUtils.cpp.o] Error 1
CMakeFiles/Makefile2:96: recipe for target 'CMakeFiles/ambf_framework.dir/all' failed
make[1]: *** [CMakeFiles/ambf_framework.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2

```
I am suspecting it having to do with the compiler C++ version (according to stackOverflow post [here](https://stackoverflow.com/questions/25594644/warning-specialization-of-template-in-different-namespace)), but am not sure where to fix it. Do you have any insights? Thanks!

adnanmunawar, 2022-08-16:
Hi Melody,

Thanks for posting the details of the issue. Are you using Ubuntu 18.04? 

Regardless, can you try out the suggestions from this [comment](https://github.com/WPI-AIM/ambf/issues/157#issuecomment-1067204924) which is from another issue but may be relevant.

melodysu83, 2022-08-16:
Thanks Adnan! 
The `ambf-2.0` branch now builds after the GCC version update. 
For anyone that has a similar issue, I updated the GCC version by running these commands in a terminal:

Check my current gcc, g++ version:
```
gcc --version
g++ --version
```

Install the gcc-7 packages:
```
sudo apt-get install -y software-properties-common
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update
sudo apt install g++-7 -y
```
Set it up so the symbolic links gcc, g++ point to the newer version:
```
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 60 \
                         --slave /usr/bin/g++ g++ /usr/bin/g++-7 
sudo update-alternatives --config gcc
gcc --version
g++ --version
```

closed 2022-08-16 by melodysu83

adnanmunawar, 2022-08-16:
Glad to hear that and thanks for the detailed info :).