WPI-AIM/ambf #187: Error message when running ambf_simulator (ambf-2.0 branch): JOINT NOT IMPLEMENTED YET

melodysu83, 2022-08-17:
Hi Adnan,

Sorry for opening another issue so quickly, but I have a quick question. I have not made modifications to any file after a fresh pull from the `ambf-2.0` branch. Upon successfully building the code and sourcing the `setup.bash` file, I ran into the "**JOINT NOT IMPLEMENTED YET**" error message whenever I import a multibody object (that contains joints) to the ambf_simulator.

The following is the output I get when typing the bare minimum command: `./ambf_simulator` (similar errors show up when I try importing raven or dvrk using the -l flag)

```
GLFW VERSION: 3.2.1 X11 GLX EGL clock_gettime /dev/js
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/launch.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./world/world.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! INITIALIZING ROS NODE HANDLE
INFO! Thread Joined: Plane
INFO! Thread Joined: light1
INFO! Thread Joined: light2
INFO! Thread Joined: default_camera
INFO! Thread Joined: World
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./multi-bodies/robots/blender-toy-car2.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! Thread Joined: Chassis
INFO! Thread Joined: ShockBL
INFO! Thread Joined: WheelBR
INFO! Thread Joined: ShockBR
INFO! Thread Joined: WheelBL
INFO! Thread Joined: ShockFL
INFO! Thread Joined: WheelFR
INFO! Thread Joined: ShockFR
INFO! Thread Joined: WheelFL
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockBL-WheelBR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockBR-WheelBL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockFL-WheelFR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockFR-WheelFL OF TYPE: JOINT NOT IMPLEMENTED YET
Segmentation fault (core dumped)
```

I feel like I must have missed something. Do you have any suggestions?
Thanks a lot!

PS1: There were similar error messages as posed in prior Issues #155 and #159, but the solutions didn't seem to apply in my case.

PS2: I also noticed the warning messages that says "**ADF version not defined thus assuming VERSION_1_0**". Is that anticipated?  Or, should I specify the ADF version in the yaml files somehow. Thanks!

adnanmunawar, 2022-08-17:
Hi Melody,

Please don't apologize for creating new issues, as a matter of fact, thank you for contributing this way.

The message `"ERROR! COMMUNICATION TYPE FOR OBJECT NAMED <name> OF TYPE: JOINT NOT IMPLEMENTED YET"` is really a warning, so don't worry about it. I shall change the word "ERROR" to "WARNING" to put people's minds at ease.

The second warning `"ADF version not defined thus assuming VERSION_1_0"` is also a warning that you can ignore. 

The core infrastructure of AMBF-2.0 is entirely different (better) than AMBF-1.0 and these warnings (/ errors) are part of that.

For this comment: 
> PS1: There were similar error messages as posed in prior Issues https://github.com/WPI-AIM/ambf/issues/155 and https://github.com/WPI-AIM/ambf/issues/159, but the solutions didn't seem to apply in my case.

Can you elaborate on what exactly is the issue as these two issues seem to be unrelated? Please feel free to open a new issue, or reopen the existing issues.

adnanmunawar, 2022-08-17:
No problem. Are you running the `ambf_simulator` with any arguments? Also, can you please provide the output of running `./ambf_simulator` in the terminal?

melodysu83, 2022-08-18:
I am not running it with any arguments. Here are the outputs I get on the terminal:
```
____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Global namespace prefix for ROS 
                                       Communication
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  - [ --plugins ] arg                  Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
                                       they are valid files
  -l [ --load_multibodies ] arg        Index of Multi-Body(ies) to Launch, 
                                       .e.g. -l 1,2,3 will load multibodies at 
                                       indexes 1,2,3. See launch.yaml file

------------------------------------------------------------

GLFW VERSION: 3.2.1 X11 GLX EGL clock_gettime /dev/js
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/launch.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./world/world.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! INITIALIZING ROS NODE HANDLE
INFO! Thread Joined: Plane
INFO! Thread Joined: light1
INFO! Thread Joined: light2
INFO! Thread Joined: default_camera
INFO! Thread Joined: World
WARNING! For File "/home/imero/Simulator/ambf/ambf_models/descriptions/./multi-bodies/robots/blender-toy-car2.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! Thread Joined: Chassis
INFO! Thread Joined: ShockBL
INFO! Thread Joined: WheelBR
INFO! Thread Joined: ShockBR
INFO! Thread Joined: WheelBL
INFO! Thread Joined: ShockFL
INFO! Thread Joined: WheelFR
INFO! Thread Joined: ShockFR
INFO! Thread Joined: WheelFL
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockBL-WheelBR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockBR-WheelBL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFL OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockFL-WheelFR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFR OF TYPE: JOINT NOT IMPLEMENTED YET
ERROR! COMMUNICATION TYPE FOR OBJECT NAMED ShockFR-WheelFL OF TYPE: JOINT NOT IMPLEMENTED YET
Segmentation fault (core dumped)
```
Also here is a quick video showing what I'm seeing (with the simulator window shutting down upon boot). In the video, I also tried loading specific robots with the -l flag, and a similar problem occurred.
https://youtu.be/D1mwX_Uw-3s
Thank you!

adnanmunawar, 2022-08-18:
Thanks for the video, and I can see that it's a segfault. We shall need to compile `AMBF` in debug mode and run it via debugger. Can you please try the following steps.

Install `gdb`(GNU Debugger) if not already installed. You would probably need `sudo` permissions
```
sudo apt install gdb
```
Then build AMBF in debug mode. Assuming that `AMBF` is located in the home `~/` folder.

```bash
cd ~/ambf
mkdir build-debug && cd build-debug
cmake .. -DCMAKE_BUILD_TYPE=Debug
make -j4
```

After a successful build, head over to the `bin-debug` folder in `AMBF`. This folder is created when `AMBF` is built in Debug mode.

```
cd ~/ambf/bin-debug/lin-x86_64
gdb ambf_simulator
```
This would launch the `gdb`  console with `ambf_simulator`. You can then enter different commands to step through the code. We shall be just running it by entering `run` and pressing `ENTER`. It should crash and when it does, could you please copy the output and paste it here.

melodysu83, 2022-08-18:
Hi Adnan,
Thanks for the tips! I tried what you suggested and here is the output I get from **gdb**.
Specifically, when I typed "**r**" to run the program, the simulator window popped up; there were nothing displayed on it, but it also didn't shut off on its own. Then to proceed with the program, I typed "**c**", that's when it shows the last three lines of outputs and shuts down the simulator window.
```
imero@IMeRo-1:~/Simulator/ambf/bin-debug/lin-x86_64$ gdb ./ambf_simulator
GNU gdb (Ubuntu 8.2-0ubuntu1~16.04.1) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./ambf_simulator...done.
(gdb) r
Starting program: /home/imero/Simulator/ambf/bin-debug/lin-x86_64/ambf_simulator 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Global namespace prefix for ROS 
                                       Communication
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  - [ --plugins ] arg                  Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
[... 265 lines cut ...]
==28499==    by 0x6A5AAF: updateGraphics() (ambf_simulator.cpp:582)
==28499==    by 0x6A5043: main (ambf_simulator.cpp:543)
==28499== 
==28499== HEAP SUMMARY:
==28499==     in use at exit: 150,709,097 bytes in 8,160 blocks
==28499==   total heap usage: 246,566 allocs, 238,406 frees, 231,460,817 bytes allocated
==28499== 
==28499== LEAK SUMMARY:
==28499==    definitely lost: 1,404 bytes in 8 blocks
==28499==    indirectly lost: 221 bytes in 5 blocks
==28499==      possibly lost: 7,849,892 bytes in 1,170 blocks
==28499==    still reachable: 142,857,580 bytes in 6,977 blocks
==28499==                       of which reachable via heuristic:
==28499==                         stdstring          : 20,568 bytes in 397 blocks
==28499==         suppressed: 0 bytes in 0 blocks
==28499== Rerun with --leak-check=full to see details of leaked memory
==28499== 
==28499== For counts of detected and suppressed errors, rerun with: -v
==28499== Use --track-origins=yes to see where uninitialised values come from
==28499== ERROR SUMMARY: 3559 errors from 7 contexts (suppressed: 0 from 0)
Killed
```
Finally, the following link shows a quick video demo: in case it helps :)
https://youtu.be/OwgHHtEviBw
Thanks for helping!!

adnanmunawar, 2022-08-18:
Awesome, great to have the valgrind output. Can you test something, if you modify this line https://github.com/WPI-AIM/ambf/blob/ambf-2.0/ambf_simulator/src/ambf_simulator.cpp#L343 and change:

```cpp
glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3);
glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 1);
```
to

```cpp
glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 2);
glfwWindowHint(GLFW_CONTEXT_VERSION_MINOR, 1);
```

and then recompile `AMBF`. Does it still crash?

melodysu83, 2022-08-18:
Wow that fixed it! Thank you so much.
Is it because of my laptop having an older version of OpenGL?

adnanmunawar, 2022-08-18:
Happy to hear that. It is probably not an issue with the OpenGL version itself as version 3.3 and greater have be supported on machines from many years ago.While reading the output from `valgrind` from your comment, I quickly realized that I accidentally pushed the line `glfwWindowHint(GLFW_CONTEXT_VERSION_MAJOR, 3)` to remote. The rendering methods need to be fixed/updated to work with the newer version of GLFW on all systems. I have not seen it crash on `20.04` but I suppose you are using `18.04`, is that correct?

melodysu83, 2022-08-18:
Ah, I see! Yes, the computers in my lab are all 18.04, but I'm currently using my ancient 16.04 laptop. I've renamed the issue title so it's not misleading and will close the comment now. Thanks again for all your help.

closed 2022-08-18 by melodysu83