WPI-AIM/ambf #202: Systemwide access to ROS modules error: no such directory as ./devel

sunshineinsandiego, 2023-04-28:
Hi - Following the instructions on the readme here: https://github.com/WPI-AIM/ambf/tree/ambf-2.0 to install on Ubuntu 20.04. Installation seemed to complete without error, but after reaching this step:

"On Linux systems, please source the correct folder to achieve system-wide availability of AMBF ROS modules. While in the build folder, you can run:"
```
source ./devel/setup.bash
```

I receive the following error:
```
$ source ./devel/setup.bash
bash: ./devel/setup.bash: No such file or directory
```

I have no such `devel` folder in my `build` directory:

```
:~/ambf/build$ ls -al
total 17048
drwxrwxr-x  6 4096 Apr 27 21:01 .
drwxrwxr-x 21 4096 Apr 27 20:57 ..
-rw-rw-r--  1 =2936 Apr 27 20:57 AMBFConfig.cmake
-rw-rw-r--  1 2053 Apr 27 20:57 AMBFConfigVersion.cmake
drwxrwxr-x  3 4096 Apr 27 20:57 ambf_simulator
-rw-rw-r--  1 3816 Apr 27 20:57 AMBFTargets.cmake
-rw-rw-r--  1 17413 Apr 27 20:57 CMakeCache.txt
drwxrwxr-x  8 4096 Apr 27 21:01 CMakeFiles
-rw-rw-r--  1 1841 Apr 27 20:57 cmake_install.cmake
drwxrwxr-x  3 4096 Apr 27 20:57 external
-rw-rw-r--  1 951374 Apr 27 21:01 libadf_loader.a
-rw-rw-r--  1 3007924 Apr 27 20:58 libambf_framework.a
-rw-rw-r--  1 4926406 Apr 27 21:01 libbullet.a
-rw-rw-r--  1 7739888 Apr 27 21:00 libchai3d.a
-rw-rw-r--  1 760625 Apr 27 20:57 Makefile
drwxrwxr-x  5 4096 Apr 27 20:57 tests
```

sunshineinsandiego, 2023-04-28:
Yes, I do have ROS in stalled on my system, but I did not source it prior to building AMBF. Didn't see that prompt on the Readme. Should I rebuild with ROS sourced?

adnanmunawar, 2023-04-28:
Yes, you can try these steps.

``` bash
cd ~/ambf/build
rm CMakeCache.txt
cmake ..
make -j8
```

sunshineinsandiego, 2023-04-28:
Sorry, just to clarif. I should try those steps in addition to sourcing ros? or try those steps alone?

adnanmunawar, 2023-04-28:
Yes, you should try these steps AFTER sourcing ROS.

sunshineinsandiego, 2023-04-28:
Will do, thanks!

sunshineinsandiego, 2023-04-28:
More errors:
```
ambf/build$ source /opt/ros/noetic/setup.bash 
ambf/build$ rm CMakeCache.txt 
ambf/build$ cmake ..
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:102 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: filesystem 
-- Using CATKIN_DEVEL_PREFIX: /home/florian/Documents/ambf/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/noetic
-- This workspace overlays: /opt/ros/noetic
-- Found PythonInterp: /home/miniconda3/bin/python3 (found suitable version "3.10.9", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /home/miniconda3/bin/python3
-- Using Debian Python package layout
-- Could NOT find PY_em (missing: PY_EM) 
CMake Error at /opt/ros/noetic/share/catkin/cmake/empy.cmake:30 (message):
  Unable to find either executable 'empy' or Python module 'em'...  try
  installing the package 'python3-empy'
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:119 (find_package)

-- Configuring incomplete, errors occurred!
See also "ambf/build/CMakeFiles/CMakeOutput.log".
```

Possibly need to install `python3-empy`?

adnanmunawar, 2023-04-28:
Hmm, I would not install it via Anaconda. Can you try the recommendations here (https://github.com/ysl208/iRoPro/issues/59)

sunshineinsandiego, 2023-04-28:
So I wanted to try using conda to practice good environment hygiene, and this is what it seems like it needs to work on my system  (for those who may run into similar problems):
```
(base) ambf/build$ conda activate ambf
(ambf) ambf/build$ source /opt/ros/noetic/setup.bash
(ambf) ambf/build$ conda install -c conda-forge empy
(ambf) ambf/build$ conda install -c conda-forge catkin_pkg
(ambf) ambf/build$ conda install -c conda-forge pyyaml
(ambf) ambf/build$ rm CMakeCache.txt
(ambf) ambf/build$ cmake ..
(ambf) ambf/build$ make -j8
...
[100%] Built target ambf_simulator
```
Build seemed to be successful. So started `roscore` in a different terminal:
```
$ source /opt/ros/noetic/setup.bash
$ roscore
```
And then returned to the original terminal:
```
(ambf) ambf/build$ source ./devel/setup.bash 
(ambf) ambf/build$ cd ../bin/lin-x86_64/
(ambf) ambf/bin/lin-x86_64$ ./ambf_simulator 
__________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Global namespace prefix for ROS 
                                       Communication
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  --plugins arg                        Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
[... 32 lines cut ...]
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockFL-WheelFR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockFR-WheelFL OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
INFO! TOTAL ACTIVE COMM INSTANCES: 14
INFO! WAITING FOR ALL COMM INSTANCES TO UNREGISTER ... 
	INFO! REMAINING ACTIVE COMMs: 14
	INFO! REMAINING ACTIVE COMMs: 5
INFO! DESTROYING ROS NODE HANDLE
INFO! Thread ShutDown: default_camera
INFO! Thread ShutDown: light1
INFO! Thread ShutDown: light2
INFO! Thread ShutDown: Chassis
INFO! Thread ShutDown: Plane
INFO! Thread ShutDown: ShockBL
INFO! Thread ShutDown: ShockBR
INFO! Thread ShutDown: ShockFL
INFO! Thread ShutDown: ShockFR
INFO! Thread ShutDown: WheelBL
INFO! Thread ShutDown: WheelBR
INFO! Thread ShutDown: WheelFL
INFO! Thread ShutDown: WheelFR
INFO! Thread ShutDown: World
```

Which seems like a good output?

adnanmunawar, 2023-05-01:
Glad you solved it. The terminal output looks good. Can you also see the GUI ?

sunshineinsandiego, 2023-05-01:
Yep, 100%. Thanks for all of your help! I'm still trying to figure out how to integrate this sim w/a docker container and ROS scripts / DVRK PSM controllers (asked the question in the discussions section [here](https://github.com/collaborative-robotics/surgical_robotics_challenge/discussions/75)), so if you have additional thoughts, would be grateful for any advice. Thanks again!

closed 2023-05-01 by sunshineinsandiego