WPI-AIM/ambf #203: Geomagic Touch connection problem

TONGLoki, 2023-05-04:
Dear Aadnanmunawar,

Hi, I have some problems about the input device. I have use [ros_geomagic] function to drive the Geomagic Touch. But my ambf simulator can’t directly find the “Geomagic Touch”. So I planned to use another method you have said: use mimic_mtms.py(https://github.com/adnanmunawar/geomagic_utils) to make a fake MTMR, then use MTMR to control the ambf. It works but not totally. It can only control the gripper button but cannot move. My project needs to control the psm and collect the poses it have moved. I don’t know how to make the mimic device become the end effector of the **PSM**. As shown in the figure, ros_gemagic is working, and can communicate with ros with location information, but after the mimic conversion, there is a problem when it is passed to ambf. Could you help me? I am using ubuntu 18.04. Please feel free to ask me any questions. I am looking forward to hear from you. Thank you!
[image]

adnanmunawar, 2023-05-04:
Hello,
Thanks for explaining what you are trying to do. If you wanted to control the PSMs via Geomagic Touch or Phantom Omni (or MTMs or Razer Hydras), I would suggest that you look at this repo (https://github.com/collaborative-robotics/surgical_robotics_challenge) and the scripts contained within. There is a Readme to help you launch the tele-operation (https://github.com/collaborative-robotics/surgical_robotics_challenge/blob/master/docs/teleoperation_supplemental.md)

TONGLoki, 2023-05-05:
Does this only work on ambf2.0? The previous version I downloaded can't run, and an error will be reported. Then I re-downloaded ambf2.0 and recompiled, and running the normal ambf_simulator also reported an error. Is this because of a conflict with the previously downloaded ambf version? How to uninstall the downloaded ambf version, is it effective to delete it directly? And this is the error:
```
tong@ub2:~/ambf/bin/lin-x86_64$ ./ambf_simulator

____________________________________________________________

ASYNCHRONOUS MULTI-BODY FRAMEWORK SIMULATOR (AMBF Simulator)

		(http://practicepoint.wpi.edu)
		  (Copyright 2019-2021)
____________________________________________________________

STARTUP COMMAND LINE OPTIONS: 

ambf_simulator Command Line Options:
  -h [ --help ]                        Show help
  -n [ --ndevs ] arg (=0)              Number of Haptic Devices to Load
  -i [ --load_devices ] arg            Index number of devices to load which is
                                       specified in input_device.yaml
  -e [ --enableforces ] arg (=0)       Enable Force Feedback on Haptic Devices
  -p [ --phx_frequency ] arg (=1000)   Physics Update Frequency (default: 1000 
                                       Hz)
  -d [ --htx_frequency ] arg (=1000)   Haptics Update Frequency (default: 1000 
                                       Hz)
  -t [ --fixed_phx_timestep ] arg (=0) Use Fixed Time-Step for Physics 
                                       (default: False)
  -f [ --fixed_htx_timestep ] arg (=0) Use Fixed Time-Step for Haptics 
                                       (default: False)
  --override_max_comm_freq arg         Override the maximum publishing 
                                       frequency for all afObjects (default: 
                                       1000 Hz)
  --override_min_comm_freq arg         Override the minimum publishing 
                                       frequency for all afObjects (default: 50
                                       Hz)
  -g [ --show_gui ] arg (=1)           Show GUI
  --ns arg                             Global namespace prefix for ROS 
                                       Communication
  -s [ --sim_speed_factor ] arg (=1)   Override the speed of "NON REAL-TIME" 
                                       simulation by a specified factor 
                                       (Default 1.0)
  - [ --plugins ] arg                  Simulator plugins to load, .e.g. 
                                       --plugins <plugin1_filepath>, 
                                       <plugin2_filepath> loads plugin1 and 
                                       plugin2 simualtor plugin
  --launch_file arg                    Launch file path to load (default: 
                                       <ROOT_PATH>/ambf_models/descriptions/lau
                                       nch.yaml
  -a [ --load_multibody_files ] arg    Description Filenames of Multi-Body(ies)
                                       to Launch, .e.g. -a <path>/test.yaml, 
                                       <another_path>/test2.yaml will load 
                                       multibodies test.yaml and test2.yaml if 
                                       they are valid files
  -l [ --load_multibodies ] arg        Index of Multi-Body(ies) to Launch, 
                                       .e.g. -l 1,2,3 will load multibodies at 
                                       indexes 1,2,3. See launch.yaml file

------------------------------------------------------------

GLFW VERSION: 3.2.1 X11 GLX EGL clock_gettime /dev/js
WARNING! For File "/home/tong/ambf/ambf_models/descriptions/launch.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/tong/ambf/ambf_models/descriptions/./world/world.yaml", ADF version not defined thus assuming VERSION_1_0
WARNING! For File "/home/tong/ambf/ambf_models/descriptions/./input_devices/input_devices.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! INITIALIZING ROS NODE HANDLE
INFO! Thread Joined: Plane
INFO! Thread Joined: light1
INFO! Thread Joined: light2
INFO! Thread Joined: default_camera
INFO! Thread Joined: World
WARNING! For File "/home/tong/ambf/ambf_models/descriptions/./multi-bodies/robots/blender-toy-car2.yaml", ADF version not defined thus assuming VERSION_1_0
INFO! Thread Joined: Chassis
INFO! Thread Joined: ShockBL
INFO! Thread Joined: WheelBR
INFO! Thread Joined: ShockBR
INFO! Thread Joined: WheelBL
INFO! Thread Joined: ShockFL
INFO! Thread Joined: WheelFR
INFO! Thread Joined: ShockFR
INFO! Thread Joined: WheelFL
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBL OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockBL-WheelBR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockBR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockBR-WheelBL OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFL OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockFL-WheelFR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED Chassis-ShockFR OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
WARNING! COMMUNICATION TYPE FOR OBJECT NAMED ShockFR-WheelFL OF TYPE: JOINT NOT IMPLEMENTED YET. IGNORING.
ERROR! PLUGIN  FAILED ON INITIALIZATION THEREFORE IGNORING!
```

adnanmunawar, 2023-05-05:
Which previous version did you download? And yes, just removed the ambf folder should be it as nothing is installed system wide.

TONGLoki, 2023-05-06:
The _feat-rl_ version. After I uninstalled(removed) feat version, I got an error after running ambf2.0: ERROR! PLUGIN FAILED ON INITIALIZATION THEREFORE IGNORING!

adnanmunawar, 2023-05-08:
I see, yeah I would just pull the latest ambf-2.0 branch, as you already did. Don't worry about that error message, it is not relevant to the surgical robotics challenge assets. Can you see the simulation with the PSMs and the Phantom ?

adnanmunawar, 2023-05-09:
Did you read this document that I attached earlier on how to launch the teleop of the PSMs? (https://github.com/collaborative-robotics/surgical_robotics_challenge/blob/master/docs/teleoperation_supplemental.md)

For questions pertaining to the surgical robotics challenge assets, please open a new discussion here (https://github.com/collaborative-robotics/surgical_robotics_challenge/discussions) as they are not directly relevant to AMBF.

TONGLoki, 2023-05-10:
Yes, I have read the document and experimented. Thanks for your help, I'll move the topic to the discussion group.