WPI-AIM/ambf #205: ambf python client mask joints up to last used idx

htp2, 2023-05-11:
Hi Adnan - as discussed today, it would be great if we could not send zero effort commands to joints after the last commanded index. This would prevent long kinematic chains from having their later joints un-controllable if commands are being sent to upstream joints/bodies. 

So if there are 8 joints numbered 1-8, and we only command up to index 6, currently, joints 7 and 8 will get sent zero effort commands. Instead, we would just not send any commands for joints 7 and 8.

Relevant section of the code is 
ambf/ambf_ros_modules/ambf_client/python/ambf_rigid_body.py Line 714.

adnanmunawar, 2023-05-12:
Hi Henry, I believe that the above commit should fix the issue. If it does, we can merge it with the master branch.

@nag92 Probably, but it seemed to no longer be the case.

adnanmunawar, 2023-05-24:
Thanks Henry, will merge it now.

closed 2023-05-24 by htp2