WPI-AIM/ambf #215: Frames are oriented differently (unit quaternion) when AMBF is launched in headless mode

kschwan, 2023-09-08:
Hi, 
I noticed that if I launch the AMBF sim with the [surgical_robotics_challenge](https://github.com/surgical-robotics-ai/surgical_robotics_challenge) environment in headless mode (`-g0`) all coordinate frames seems to be oriented differently (ie., with unit orientation). See below for an example.

#### In normal, non-headless, mode
Terminal 1:
```
kils@x:~$ ambf_simulator --launch_file ~/source/surgical_robotics_challenge/launch.yaml -l 0,1,3,4,14,17 -p120 -t1 --override_max_comm_freq 120
```

Terminal 2:
```
kils@x:~$ rostopic echo -n1 /ambf/env/phantom_001/State
header: 
  seq: 76
  stamp: 
    secs: 1694176163
    nsecs: 109043598
  frame_id: ''
sim_step: 116
identifier: 
  data: "/ambf/env/BODY phantom_001"
name: 
  data: "phantom_001"
wall_time: 1.5306248664855957
sim_time: 0.9666666388511658
mass: 0.0
pInertia: 
  x: 0.0
  y: 0.0
  z: 0.0
pose: 
  position: 
    x: -0.0194744784
    y: 0.1155202687
    z: 0.6571981311
  orientation: 
    x: -0.48321465615355697
    y: 0.5162399418206148
    z: -0.5162397017891298
    w: 0.4832147440260064
twist: 
  linear: 
    x: -1.764430404428116e-05
    y: 1.4361735273925265e-05
    z: -1.3985269249818563e-13
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
wrench: 
  force: 
    x: 0.0
    y: 0.0
    z: 0.0
  torque: 
    x: 0.0
    y: 0.0
    z: 0.0
children_names: []
joint_names: []
joint_positions: []
joint_velocities: []
joint_efforts: []
---
```

#### In headless mode
Terminal 1:
```
kils@x:~$ ambf_simulator --launch_file ~/source/surgical_robotics_challenge/launch.yaml -l 0,1,3,4,14,17 -p120 -t1 --override_max_comm_freq 120 -g0
```

Terminal 2:
```
kils@tek-cb-kils-02:~$ rostopic echo -n1 /ambf/env/phantom_001/State
header: 
  seq: 69
  stamp: 
    secs: 1694176277
    nsecs: 941417933
  frame_id: ''
sim_step: 164
identifier: 
  data: "/ambf/env/BODY phantom_001"
name: 
  data: "phantom_001"
wall_time: 1.3850514888763428
sim_time: 1.3666666746139526
mass: 0.0
pInertia: 
  x: 0.0
  y: 0.0
  z: 0.0
pose: 
  position: 
    x: -0.0194744784
    y: 0.1155202687
    z: 0.6571981311
  orientation: 
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
twist: 
  linear: 
    x: 0.0
    y: 0.0
    z: 0.0
  angular: 
    x: 0.0
    y: 0.0
    z: 0.0
wrench: 
  force: 
    x: 0.0
    y: 0.0
    z: 0.0
  torque: 
    x: 0.0
    y: 0.0
    z: 0.0
children_names: []
joint_names: []
joint_positions: []
joint_velocities: []
joint_efforts: []
---
```

kschwan, 2023-09-08:
Not sure if this is an AMBF bug per se but I assumed it is. I didn't try launching other simulator configurations / environments.

adnanmunawar, 2023-09-26:
Hi Kim,

Thanks for explaining the issue. Can you pull the latest commit (above) and recompile AMBF to see if the issue still persists?