WPI-AIM/ambf #218: Can't import ambf_client

darthandvader, 2023-10-14:
I've sourced the ambf/build/devel folder and can find the path in my bash file but still when I use the python script in pycharm htere's still "No module named ambf_client". I also tried switching to python 2.7 interpreter but it didn't work.

I looked at the issue #9 and #67 it seems like they can launch the client after adding the path so I'm comfused which step I missed.

Thank you!

adnanmunawar, 2023-10-18:
Hello, 

Are you running Pycharm via the terminal with your `.bashrc` file sourced?

darthandvader, 2023-10-19:
Hi, I run Pycharm directly from application interface. Should I run it from terminal instead?

adnanmunawar, 2023-10-19:
Yes, you should run it from a terminal with the ROS and AMBF sourced prior to launch.