WPI-AIM/ambf #239: Python import statements for ambf_comm, ambf_world, etc.

juseonghan, 2024-11-08:
Hello - we are trying to create an AMBF wrapper to test and deploy RL algorithms, and we found your `AmbfEnv()` class that you specified in `ambf_env.py`. However, if I run `env_test.py` I get import errors. Upon closer inspection, it seems like import statements generally fail if it's not from `ambf_client` or `ambf_msgs` (e.g. `import ambf_world`, `import ambf_comm` all do not work). Is there a fix for this or are we doing something incorrectly from our side? We have ROS and AMBF sourced, and the client works fine otherwise. Thanks!

adnanmunawar, 2024-11-12:
Hello, it has been a while since we used this functionality, please try updating your repo to include the commits above and check if this fixes the issue.

Broadly, you may want to check out this repo for our current integration with gym https://github.com/surgical-robotics-ai/SurgicAI

adnanmunawar, 2026-03-13:
Closing this as stale issue.