WPI-AIM/ambf #242: arm joints constraint and collision constraint conflict

SNEAndy, 2025-02-28:
Hi, I have two robot arms in my scene,which include clips and tool-heads. I use hinge constraint to connect the clips with its tool-head, it works well. But when i control these two arms to  collide with each other，their joints disintegrated. I tried to fix that but I got no clue.
can you help me ?

adnanmunawar, 2025-03-25:
Hello, I will have to look at the model files. Did you use [blender-ambf addon](https://github.com/WPI-AIM/ambf_addon) to model your robots? If so, have you tried playing around with the collision configuration of the rigid bodies?

-Adnan

SNEAndy, 2025-03-26:
Hi,Adnan,thanks for your reply.yes, i did use the blender-ambf addon to model my robots,here is my robot yaml file: [robotArm1.txt](https://github.com/user-attachments/files/19459380/robotArm1.txt)
sorry that i can't provide OBJ file of my robot,i'm not allowed to do that. 
I set the angle of  every joint,then used the  `commandPosition` to set the target Motor velocity of every joint to control them. When i control arm to collide another rigid body,like another arm , the ground or a fixed rigid body, the joints of my arm disintegrated,but if i control the arm to move away, the joints recover. Is there any way I can stop the arm or limit it when arm collides? 
hope your reply

adnanmunawar, 2025-03-27:
No problem. I see that you use CONVEX_HULLs and CONVEX_MESHes for the collision, which is good. Here are two points to consider.
1. The PD values of the joint controllers are very high. I would lower them significantly. Ideally P <= 10, and D <= 0.1
2. Inertial offset of ToolBody is large. This could cause issues. Try to balance mesh origin to the previous joint location and set inertial offset closer to it.
3. Make sure that the resolution of your collision meshes is reasonable. Ideally under 500 vertices.
4. The masses are really small, try increasing them proportionally to see if the behavior improves.

SNEAndy, 2025-03-28:
> No problem. I see that you use CONVEX_HULLs and CONVEX_MESHes for the collision, which is good. Here are two points to consider.
>
> 1. The PD values of the joint controllers are very high. I would [...]

yes, you‘re absolutely right! I tried to increase the mass of every rigid body and the problem is solved. Thank you very much, you did me a good favor! hope you have a good day!

adnanmunawar, 2025-03-28:
Glad to hear. I shall convert this issue into a discussion.